openmiko/openmiko #140: SSH Connection Refused

ndrwstn, 2024-08-15:
**Describe the bug**
Using `v1.0.0-alpha.1` on Wyze Cam V2 with `openmiko.conf` and `wpa_supplicant` created from `utilities`, ssh is refusing connections on port 22.

**To Reproduce**
1. Reflash with `demo.bin`, `openmiko.conf` and `wpa_supplicant`
2. `ssh -vvv <ip>`
```
OpenSSH_9.7p1, LibreSSL 3.3.6
debug1: Reading configuration data /Users/austin/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: include /etc/ssh/ssh_config.d/* matched no files
debug1: /etc/ssh/ssh_config line 54: Applying options for *
debug1: Authenticator provider $SSH_SK_PROVIDER did not resolve; disabling
debug1: Connecting to <ip> [<ip>] port 22.
debug1: connect to address <ip> port 22: Connection refused
ssh: connect to host <ip> port 22: Connection refused
```

**Expected behavior**
Able to connect to SSH.

**Additional Notes**
I was slowly stripping away custom overlay files in an attempt to figure out why ssh was gorked, as another issue #48 suggested it was a malformed json file. I finally got back to bone-stock re-flash and it still isn't connecting. However, I did note that all of my custom overlays were working correctly[1] earlier (I just couldn't ssh in--so I couldn't verify `passwd`/`shadow`).

[1] TZ wasn't working as expected, but it did offset the time the amount I input, it was just the wrong offset. I assume that's something wrong I did and unrelated to the above issue.

martinbutt, 2024-08-15:
Could be the same issue as https://github.com/openmiko/openmiko/issues/130

[v0.0.40](https://github.com/openmiko/openmiko/releases/tag/v0.0.40) is the last stable version, but note that the config file formats are different, so it is best to start from scratch. You can also try [v1.0.1-alpha.1](https://github.com/martinbutt/openmiko/releases/tag/v1.0.1-alpha.1), which has some fixes in it.

ndrwstn, 2024-08-17:
SSH is still refused with [v1.0.1-alpha.1](https://github.com/martinbutt/openmiko/releases/tag/v1.0.1-alpha.1). Unfortunately, I am having the same problem with [v.0.0.40](https://github.com/openmiko/openmiko/releases/tag/v0.0.40). I have two other older Wyze Cameras that I successfully flashed (though have not tried to re-flash, since they currently work). I went back to the release I think they were on ([v0.0.28](https://github.com/openmiko/openmiko/releases/tag/v0.0.28)) and tried to flash that: same problem, even with only `wpa_supplicant.conf'.

I'm starting to think this is a "me problem" but I don't know what to troubleshoot.

martinbutt, 2024-08-17:
One suggestion when things like this happen is to ensure that you are fully resetting the persistent storage. Turn on the camera and after is initialized, hold down the reset button for 5 to 10 seconds. Then try a firmware from scratch. Another problem can be a corrupted SD card. Run fsck on it using a different computer, and make sure it's cleanly unmounted before plugging it into the camera.

ndrwstn, 2024-09-01:
So it took a while to get back to playing with this, but your suggestion (reset persistent storage) appeared to work. I re-flashed with 0.0.40 and I'm able to login. Haven't played with anything else, but I'm going to close this issue.

Many thanks for your assistance @martinbutt .

closed 2024-09-01 by ndrwstn