openmiko/openmiko #145: Night mode is bouncing multiple times a minute

eastoncrafter, 2025-04-06:
**Describe the bug**
I've got a Wyze v2 camera watching a cat door, and every night, when I turn down the lights, the camera doesn't seem to be able to make up its mind on whether to be in night mode or day mode. The stream will bounce between night and day every 30 seconds or so. Is there a way to tune the night/day filter somehow?

**Specs**
Wyze V2 ([0.0.41-alpha.2](https://github.com/openmiko/openmiko/releases/tag/v0.0.41-alpha.2)) I think

**To Reproduce**
Steps to reproduce the behavior:
Take the camera into some place with low light
The stream will bounce every 30 seconds (about)

**Expected behavior**
The camera should stay in one mode or the other for low light, without bouncing.

Can upload a video if necessary

martinbutt, 2025-04-06:
Unfortunately this project is no longer maintained. This seems to be the most active alternative http://www.thingino.com/

eastoncrafter, 2025-05-12:
> Unfortunately this project is no longer maintained. This seems to be the most active alternative http://www.thingino.com/

I attempted flashing, but I'm still on openmiko for some reason. Do I need to flash back to stock firmware first?